jmccrohan/pysolarmanv5 #3: unable to install on raspberry pi

Woodpeckercz, 2022-04-08:
Hi,

I'm getting an error when I try to install this on my raspberry pi. Any advice?
thanks.

_pip install .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/pysolarmanv5
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-t1Arv7/setup.py", line 3, in <module>
        with open("README.md", "r", encoding="utf-8") as fh:
    TypeError: 'encoding' is an invalid keyword argument for this function_

Woodpeckercz, 2022-04-08:
when I try to install using _pip install pysolarmanv5_ i get this:

_pip install pysolarmanv5
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pysolarmanv5
  Could not find a version that satisfies the requirement pysolarmanv5 (from versions: )
No matching distribution found for pysolarmanv5
_

jmccrohan, 2022-04-20:
No response. Closing.

closed 2022-04-20 by jmccrohan