jmccrohan/pysolarmanv5 #6: raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")

dcelorrio, 2022-05-23:
With a deye sun hybrid 8000w , I execute client_example.py and finish with this error . (Device serial number 2330702165 and Inverter serial number 2201144460)

What is the problem?

Thanks

SENT: a5 17 00 10 45 00 00 55 b1 eb 8a 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 80 fe 00 06 38 38 e2 15
RECD: a5 10 00 10 15 00 0d 55 b1 eb 8a 02 01 75 b8 06 00 c2 02 00 00 21 eb 84 62 05 00 ae 15
Traceback (most recent call last):
  File "client_example.py", line 44, in <module>
    main()
  File "client_example.py", line 17, in main
    print(modbus.read_input_registers(register_addr=33022, quantity=6))
  File "/home/dcelorrio/myapp/my_app_venv/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 224, in read_input_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "/home/dcelorrio/myapp/my_app_venv/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 178, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "/home/dcelorrio/myapp/my_app_venv/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 173, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/home/dcelorrio/myapp/my_app_venv/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 153, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame

jmccrohan, 2022-05-31:
Hi @dcelorrio,

The frame you have received, while a valid solarmanv5 frame, doesn not contain a valid Modbus RTU frame.

Have you amended the client example registers to match your Deye Sun Hybrid 8000w device? It will most likely have different Modbus registers.

The `client_example.py` file is just there to demonstrate how to use the library.

Regards,
Jon

closed 2022-06-17 by jmccrohan

Xhanti, 2022-09-02:
See my answer here:

https://github.com/jmccrohan/pysolarmanv5/issues/12