jmccrohan/pysolarmanv5 #7: Ethernet cable connected device

drsmarsden, 2022-05-30:
Hi

I have just started experimenting with this. So far:

1) the unit is discoverable
2) does not allow a remotely initiated connection
3) does allow a second server to be specified in the web UI to allow both servers to be updated
4) I have done a hack to allow the device to do the connection
5) it will initiate a TCP connection and send periodic data and keep alive packets
6) I am struggling to get meaningful data for some values I am looking for - it is also not obvious what is in the periodic data

I am now retired for IT management, so only hobby developer!
any help appreciated - ultimate destination for the data is Home Assistant

Stuart

jmccrohan, 2022-05-31:
Hi @drsmarsden,

I am using this with a Wifi data logging stick, which does listen on tcp/8899 by default. I don't have an Ethernet data logging stick so I'm not sure how much assistance I can be unfortunately.

If I navigate to the URL below, I see the config screen below. You can see port 8899 defined under Internal Server Parameters Setting
http://<DATA_LOGGER_IP_ADDRESS>/config_hide.html

[image]

drsmarsden, 2022-06-01:
Hi

Thanks

Weird - if I use your original code I can get a TCP connection to the logger on port 8899 , send a request but get no reply

If I add my PC as server B on the logger, change the code so we just listed on port 10004 we get a TCP connection to us and receive data (but so far only the date looks correct)

I am trying to get the API keys from Ginlong so I can try the cloud route

thanks

Stuart

RaoulSargent, 2022-06-24:
Hi Stuart.

Because Ginlong are moving people to a new portal I am in the process of moving away from pulling the data from the web API and instead pulling directly from the data logger (no internet connection required) - however I saw your comment and thought I would share what I have been using to pull the data via the Ginlong v1 portal ... Hope it helps you.

I have been using the following code for the past year - it polls the Ginlong API that are called by the Ginlong V1 portal web pages.  In other words it pretends to be 'you', logging in as 'you' and grabbing the JSON responses and then posts them to a MQTT server so I can process with NodeRed/etc

I have a NodeRed node node that calls this python script on a 5 min interval.
I chose to do this, rather than have the python script loop/repeat so that I can dynamically control the time interval from NodeRed.

If you don't want to run this code, the URL's that you need are contained in the code below.

**ginlong_scraper_rs_norepeat.py**

_Note: I did **not** write the original code, and I massively apologise to the original author as I have forgotten his name, sorry._

_Note: Mentions of 'Raoul', 'Toby' or 'RS_' are my own additions/comments ... sorry I have left in the debugging, it should be cleaned up._

```
#!/usr/bin/python
import requests
import urllib
import json
import datetime
import time
import os
import logging
import schedule

# Not all keys are avilable depending on your setup
COLLECTED_DATA = {
    'DC_Voltage_PV1': '1a',
    'DC_Voltage_PV2': '1b',
    'DC_Current1': '1j',
    'DC_Current2': '1k',
    'AC_Voltage': '1ah',
    'AC_Current': '1ak',
    'AC_Power': '1ao',
    'AC_Frequency': '1ar',
    'DC_Power_PV1': '1s',
    'DC_Power_PV2': '1t',
    'Inverter_Temperature': '1df',
    'Daily_Generation': '1bd',
    'Monthly_Generation': '1be',
    'Annual_Generation': '1bf',
    'Total_Generation': '1bc',
    'Generation_Last_Month': '1ru',
    'Power_Grid_Total_Power': '1bq',
    'Total_On_grid_Generation': '1bu',
    'Total_Energy_Purchased': '1bv',
    'Consumption_Power': '1cj',
    'Consumption_Energy': '1cn',
    'Daily_Energy_Used': '1co',
    'Monthly_Energy_Used': '1cp',
    'Annual_Energy_Used': '1cq',
    'Battery_Charge_Percent': '1cv'
}

def do_work():
    # solis/ginlong portal config

    # Raoul 2021-06-04
    #username = os.environ['GINLONG_USERNAME']
    #password = os.environ['GINLONG_PASSWORD']
    #domain = os.environ['GINLONG_DOMAIN']
    #lan = os.environ['GINLONG_LANG']
    #deviceId = os.environ['GINLONG_DEVICE_ID']

    username = 'PUTYOURLOGINHERE'
    password = 'PUTYOURPASSWORDHERE'
    domain = 'm.ginlong.com'
[... 228 lines cut ...]
get_loglevel = 'DEBUG'
# Raoul 2021-06-04

loglevel = logging.INFO
if get_loglevel.lower() == "info":
    loglevel = logging.INFO
elif get_loglevel.lower() == "error":
    loglevel = logging.ERROR
elif get_loglevel.lower() == "debug":
    loglevel = logging.DEBUG

logging.basicConfig(level=loglevel, format='%(asctime)s %(levelname)s %(message)s')
logging.info('Started ginlong-solis-scraper')

#schedule.every(4).minutes.at(':00').do(main).run()
#while True:
#    if next_run_yes == 1:
#        next_run = schedule.next_run().strftime('%d/%m/%Y %H:%M:%S')
#        logging.info('Next run is scheduled at %s' % next_run)
#        next_run_yes = 0
#    schedule.run_pending()
#    time.sleep(1)

main()
```

drsmarsden, 2022-06-26:
Hi
Thanks - I will give this a go
Still no progress on getting a solis cloud api key

Stuart

drsmarsden, 2022-07-19:
> Hi @drsmarsden <https://github.com/drsmarsden>,
>
> I'm doing a bit of housekeeping on older issues. Did you ever manage to get pysolarmanv5 working with your ethernet logger?
>
> Regards,
> Jon

Hi

The wifi logger will not accept connections. Instead,  on the logger web interface it is possible to add an additional “user” server. The logger will then make a connection to pysolarmanv5

In addition to replies to data requests the logger will periodically send unsolicited data, and short “keepalive?” Packet.

I had real issues trying to get sensible data and Ginlong refused to provide packet data.

I finally get access to the soliscloud.com API and I am testing https://github.com/hultenvp/solis-sensor <https://github.com/hultenvp/solis-sensor>

The data into HA appears to be mostly correct but the "Energy dashboard” is giving mostly wrong results, which I am starting to investigate.

Thanks for all your help

Stuart

jmccrohan, 2022-07-19:
Hi @drsmarsden,

Thanks, that is very helpful to know. Does this provide full Modbus RTU access using pysolarmanv5?

Would you mind documenting the steps for the additional user server and I can add to the documentation (with accreditation of course!). 

Regards,
Jon

jmccrohan, 2022-09-14:
Closing; Duplicate of #5