jmccrohan/pysolarmanv5 #8: Solis branded stick with serial number that contains letters - cannot connect

cn-irl, 2022-06-04:
Hi,

Thanks for the great work on this project.

I am having trouble dealing with my device. I have a wifi data logging stick that seems to have undergone some sort of Solis rebrand. 

The firmware version is 0001012F and my device serial number has 16 characters and contains a mixture of letters and numbers. I still have the box it came in and the QR code on the outside gives the same serial number but also another 10 digit number that starts with zero.

Having tried every combination of these numbers I can think of I am constantly either getting a syntax error due to letters in the serial number of a Error 111 connection refused.

I also cannot navigate to http://ipaddress/config_hide.html - the page doesn't seem to exist on my unit.

I'd appreciate any insight you might have - although i know what I'm describing is not much to go on!!

Thanks

huishuis, 2022-06-19:
I have the same data logging stick with the same serial number. Were you able to log onto the wifi stick by navigating to the IP adress and using admin/admin  as username/password or admin/123456789 ? This has been suggested in a few places to log onto the data logging stick, but does not work for me.

cn-irl, 2022-06-19:
I had a really weird experience with this one. I'd love to know if it is replicated on your end ....

After countless hours of randomly typing in combinations I was eventually able to log in with:
username:  admin
password:   *the_password_of_my_home_wifi*

I have no idea whether the installer accidentally managed to set the unit password to be the same as my wifi or whether it's some snafu in the set up process. If you have the same installer it may even work for you!

huishuis, 2022-06-20:
Using the wifi password allowed me to log onto the data logger. But similar to you, the config_hide page does not exist. 
I also have the same firmware version and also the mix of letters and numbers as the serial number. 

The ‘date  of factory’ of the data logger mentioned in SolisCloud is 2022-05-17.

jmccrohan, 2022-06-20:
Any hints within the HTML source code for a hidden config page etc?

jmccrohan, 2022-06-20:
@cn-irl I've added a minimal Solarman scanner as part of 2391ec4. It would be worth running this to confirm 100% that the S3-WIFI-ST is not Solarman based.

When running the utility on a network with Solarman data loggers, you should receive the following output:

```
user@host:~/src/pysolarmanv5$ python utils/solarman_scan.py 
{'ipaddress': '192.168.1.10', 'mac': 'ABCDEF123456', 'serial': '4051234567'}
```

cn-irl, 2022-06-22:
@jmccrohan  Nothing at all when I run that script. Just a second or two of silence and then I see the command line again

The logger is on a different subnet to the device on which I'm running the script. Does that matter?

Also - to answer another question there appear to be one or two hidden html pages on the stick. There's one to set the IP address of the server and another to upload new firmware. However - having tried to add a second local server address absolutely nothing happens so I suspect it could be a legacy page that wasn't purged rather than a page that's intended for use.

I'm by no means expert in the area but by opening the hidden page's source code and adding in stop points I was able to watch the script running that should update some array with the newly server address it didn't seem to lead anywhere. This is a week or two ago so I can't remember the specifics. 

My next thought was that I could somehow trace the storage location on the device of the array that stores the server address and manually input the local address I want to use ...... at that point I ran out of both time and ability. 

I've had to park the whole thing for the last week or two due to lack of time.

jmccrohan, 2022-06-22:
@cn-irl The scan script sends a udp broadcast packet which is, by definition, limited to the current subnet. If your logger is on a different subnet, it will never receive the broadcast packet. You'll have to either temporarily join your IoT VLAN and retry it, or else use something like [udp-broadcast-relay](https://github.com/udp-redux/udp-broadcast-relay-redux) to relay the broadcast packet to the IoT subnet. I suspect the former is the quicker option though. :-)

cn-irl, 2022-06-23:
No luck unfortunately. Total silence when I run the script on the same subnet as the logger.

OneScuzzie2k, 2022-06-25:
Has anyone had any joy with the Gen3 wifi stick? I have got a SOLIS-DLS-WIFI as I cannot get the S3-WIFI-ST to work

jmccrohan, 2022-07-01:
After a bit of research, I've determined that the Solis S3-WIFI-ST uses MXCHIP EMW3080-E SOCs, rather than the Hi-Flying HF-A11 SOCs used in Solarman based loggers.

Someone might be able figure out how to configure the MXCHIP EMW3080-E using AT commands to pass through raw commands to the serial port, but that is beyond the scope of this library.

closed 2022-07-01 by jmccrohan

hn, 2025-05-11:
[This project](https://github.com/hn/ginlong-solis) has more details on the Solis S3 stick, the MXCHIP EMW3080 and how to read from (or replace) the firmware.