jmccrohan/pysolarmanv5 #9: Help understanding Working Status

LucidityCrash, 2022-06-09:
More a request for info than a bug :)
I found this project through https://github.com/StephanJoubert/home_assistant_solarman and while that was working well I was trying to understand the Working Status
I've looked at the Modbus doc from here : https://community.openenergymonitor.org/uploads/short-url/860gETtHwhFvy2a9AiIzDqO2lT5.pdf
and using your example code as a base
```
    modbus = PySolarmanV5(
        "192.168.1.21", 408xxxxxxx, port=8899, mb_slave_id=1, verbose=1
    )

    print(modbus.read_input_registers(register_addr=33121, quantity=1))
```
I get an output of 
```
SENT: a5 17 00 10 45 00 00 71 19 8e f3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 81 61 00 01 48 28 d1 15
RECD: a5 15 00 10 15 00 c2 71 19 8e f3 02 01 e2 ca 0b 00 a9 25 00 00 c1 36 96 62 01 04 02 07 01 7a c0 c7 15
[1793]
```
1793 is 0x701 which is what I get in the afformentioned Home Assistant Integration
both of these translate to 0000 0111 0000 0001 which according to the appendix would suggest that there is a Load Failure, Grid Failure, and Battery Failure.  None of which seem to be indicated, in the web UI at least - not currently in front of the inverter.

Any thoughts ?

LucidityCrash, 2022-06-09:
> Possibly the wrong spec for your inverter, Solis doesn't have a standard that works over all of their inverters, seems crazy, but that's how it is. I found a close match for my inverter and noticed that it says you have to read register_addr-1 to get get the value for register_addr....

I've got a RHI-3-6K-48ES-5G which I think should be covered by the PDF that I linked and that register also lines up with the yaml decode @jmccrohan wrote for the Home Assistant Integration I linked which mentions this inverter  in the comments :grin:

jmccrohan, 2022-06-11:
Hi @LucidityCrash,

I've noticed this as well. During normal operation, register `33121` on my `RHI-3-6K-48ES-5G` contains a value of `0x701` (`1793` in decimal). I believe bits 8, 9 and 10 are ordinarily `1` and change to `0` when a load. grid, and battery fault occurs respectively.

The reason for this theory is that the same bits are also individually addressable using `read_discrete_inputs()` (Modbus function code 0x02)
```
    modbus = PySolarmanV5("192.168.1.21", 408xxxxxxx)

    print(modbus.read_input_registers(register_addr=33121, quantity=1))
    print(modbus.read_discrete_inputs(register_addr=12589, quantity=3))
```

The same Modbus doc you have linked above describes these 12xxx registers as status registers rather than fault registers.

Hope this helps!

Regards,
Jon

jmccrohan, 2022-06-11:
> Possibly the wrong spec for your inverter, Solis doesn't have a standard that works over all of their inverters, seems crazy, but that's how it is. I found a close match for my inverter and noticed that it says you have to read register_addr-1 to get get the value for register_addr....
> 

@gedger This is a Modbus thing rather than a Solis thing. See this for a quick explanation: https://www.youtube.com/watch?v=_JzMpYxkbDU

LucidityCrash, 2022-06-12:
That is certainly a reasonable explination.
Thanks

closed 2022-06-12 by LucidityCrash