jmccrohan/pysolarmanv5 #12: Modbus RTU frame error with Deye Inverter and Solarman LSW3 stick logger

VIP-code, 2022-06-26:
Hi 

I am attempting to use pysolarmanv5 with a Deye Inverter and a Solarman LSW 3 stick logger (wifi). 

I can connect to the logger with pysolarmanv5. A request is sent and a response is received, like the following:

```
SENT: a5 17 00 10 45 00 00 dd bc f4 6a 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 80 fe 00 06 38 38 5e 15
RECD: a5 10 00 10 15 00 40 dd bc f4 6a 02 01 ae 38 11 00 57 0e 00 00 7b 5b a7 62 05 00 af 15
```

However the following exception is then thrown

```
Traceback (most recent call last):
  File "/Users/vp/pysolarmanv5-master/examples/client_example.py", line 44, in <module>
    main()
  File "/Users/vp/pysolarmanv5-master/examples/client_example.py", line 17, in main
    print(modbus.read_input_registers(register_addr=33022, quantity=6))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pysolarmanv5/pysolarmanv5.py", line 224, in read_input_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pysolarmanv5/pysolarmanv5.py", line 178, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pysolarmanv5/pysolarmanv5.py", line 173, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pysolarmanv5/pysolarmanv5.py", line 153, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame
```

Please let me know if you have any recommendations to fix the above.

Thank You.

jmccrohan, 2022-06-27:
Hi @VIP-code,

The registers used in `client-example.py` are based on a Solis Hybrid inverter. You will need to amend to suit your Deye inverter.

Regards,
Jon

closed 2022-06-27 by jmccrohan

Xhanti, 2022-09-02:
Hi,

You need to look at the holding registers not the input registers like in the sample. Here is a link to the relevant registers for the Deye: https://powerforum.co.za/topic/8451-sunsynk-inverter-monitoring/?do=findComment&comment=100858. It says Sunsynk, but the registers are largely the same for Deye(since Deye is the OEM for sunsynk). Please be careful as you can both read and write registers  on the Deye using this library and you need to know what you doing if you plan to write to registers. This library rocks by the way.