jmccrohan/pysolarmanv5 #14: V5 frame contains incorrect data logger serial number

Bonnee, 2022-07-06:
Hi, I'm trying to connect to my solarman micro inverter, but I get the following output when running the examples/client_example.py
```
SENT: a5 17 00 10 45 00 00 cb 4c d5 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 80 fe 00 06 38 38 d0 15
RECD: a5 10 00 10 15 00 04 37 5c 79 f2 02 01 f3 6d 04 00 64 00 00 00 00 00 00 00 06 00 08 15
Traceback (most recent call last):
  File "/home/bonnee/src/pysolarmanv5/examples/client_example.py", line 44, in <module>
    main()
  File "/home/bonnee/src/pysolarmanv5/examples/client_example.py", line 17, in main
    print(modbus.read_input_registers(register_addr=33022, quantity=6))
  File "/home/bonnee/.local/lib/python3.10/site-packages/pysolarmanv5/pysolarmanv5.py", line 224, in read_input_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "/home/bonnee/.local/lib/python3.10/site-packages/pysolarmanv5/pysolarmanv5.py", line 178, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "/home/bonnee/.local/lib/python3.10/site-packages/pysolarmanv5/pysolarmanv5.py", line 173, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/home/bonnee/.local/lib/python3.10/site-packages/pysolarmanv5/pysolarmanv5.py", line 144, in _v5_frame_decoder
    raise V5FrameError("V5 frame contains incorrect data logger serial number")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame contains incorrect data logger serial number
```
I have set the IP and S/N according to the inverter's web view (S/N is 211XXXXXXX)

jmccrohan, 2022-07-06:
Hi @Bonnee,

Are you sure that you are using the correct S/N (data logging stick and NOT the inverter)?

Based on the reply frame, I would expect your data logging stick serial number to be 406803xxxx. If you are in doubt, use the [solarman_scan.py](https://github.com/jmccrohan/pysolarmanv5/blob/main/utils/solarman_scan.py) utility, which will provide you with the IP address, MAC address and S/N of your data logging stick.

Regards,
Jon

Bonnee, 2022-07-06:
I don't have a stick, I'm connecting to the inverter directly. The solarman_scan utility does not seem to find my inverter.
I know the inverter works because I can get data through the homeassistant integration

jmccrohan, 2022-07-19:
As per discussion with @Bonnee via email;
Incorrect data logging serial number was used.

Communication issues resolved once correct data logging stick serial number was used, but microinverter doesn't appear to support Modbus.

closed 2022-07-19 by jmccrohan