jmccrohan/pysolarmanv5 #16: Solis DLS-W not working

ishanfdo18098, 2022-08-05:
Inverter - Solis S6-GR1P4.6K
Datalogger - DLS-W 
[image]

I am currently registered in the [soliscloud](https://www.soliscloud.com/) website and the [solarman](https://home.solarman.cn/main.html) website and both work perfectly.

I tried the utils/solarman_scan.py script and it doesn't output anything

```bash
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/utils$ python3 solarman_scan.py 
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/utils$ 
```

Datalogger site
[image]
http://datalogger.ip/config_hide.html works,
[image]

Ran the register_scan.py script and it shows,
```
Scanning input registers
Register: 35000         Value: 01010 (0x03f2)
Traceback (most recent call last):
  File "C:\Users\IshanFernando\Desktop\tmp\pysolarmanv5\examples\register_scan.py", line 31, in <module>
    main()
  File "C:\Users\IshanFernando\Desktop\tmp\pysolarmanv5\examples\register_scan.py", line 14, in main
    val = modbus.read_input_registers(register_addr=x, quantity=1)[0]
  File "C:\Users\IshanFernando\AppData\Local\Programs\Python\Python39\lib\site-packages\pysolarmanv5\pysolarmanv5.py", line 347, in read_input_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "C:\Users\IshanFernando\AppData\Local\Programs\Python\Python39\lib\site-packages\pysolarmanv5\pysolarmanv5.py", line 261, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "C:\Users\IshanFernando\AppData\Local\Programs\Python\Python39\lib\site-packages\pysolarmanv5\pysolarmanv5.py", line 248, in _send_receive_modbus_frame
    v5_response_frame = self._send_receive_v5_frame(v5_request_frame)
  File "C:\Users\IshanFernando\AppData\Local\Programs\Python\Python39\lib\site-packages\pysolarmanv5\pysolarmanv5.py", line 233, in _send_receive_v5_frame
    v5_response = self.sock.recv(1024)
socket.timeout: timed out
```

Ran the first print() in the client_example.py and it shows the 1010 value
```bash
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$ python3 client_example.py 
[1010]
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$
```

But I feel like the value doesnt represent anything.

Any help will be much appreciated with my issue.

gedger, 2022-08-05:
On the config_hide.html page did you see the Internal Server setting? If this is missing then I suspect your dongle isn't compatible.

[image]

ishanfdo18098, 2022-08-05:
I can see the same options in the config page
[image]

ishanfdo18098, 2022-08-05:
Do I have to select anything for the mode ?
[image]

gedger, 2022-08-05:
You don't need to change any parameters in the inverter settings, you edit the register_scan.py file and enter the IP address of your inverter, the serial number of the Device (not the inverter) and the port of 8899.

I've just checked and your dongle has identical firmware to mine so it will work.

Double check you can ping the inverter from the machine running the python code to ensure the IP connection is OK.

ishanfdo18098, 2022-08-05:
> you edit the register_scan.py file and enter the IP address of your inverter, the serial number of the Device (not the inverter) and the port of 8899.

Yes. In examples/register_scan.py line 8 I changed the IP to the dataloggers IP, and the serial number to the number written in the datalogger. (The first image in opening post. starting with 409......... 10 digits  )

> Double check you can ping the inverter from the machine running the python code to ensure the IP connection is OK.

ping 
```bash
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$ ping 192.168.1.103
PING 192.168.1.103 (192.168.1.103) 56(84) bytes of data.
64 bytes from 192.168.1.103: icmp_seq=1 ttl=255 time=82.2 ms
64 bytes from 192.168.1.103: icmp_seq=2 ttl=255 time=4.30 ms
64 bytes from 192.168.1.103: icmp_seq=3 ttl=255 time=6.41 ms
^C
--- 192.168.1.103 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 4.298/30.958/82.168/36.221 ms
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$ ^C
```

changed the verbose to True in the example py.  There I see that the datalogger sends a reply.
``` 
modbus = PySolarmanV5(
        "192.168.1.103", 4095xxxxxx, port=8899, mb_slave_id=1, verbose=True
    )
```

``` bash
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$ python3 client_example.py
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 00 00 a0 83 20 f4 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 88 b8 00 01 9a 4f d4 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 00 1c a0 83 20 f4 02 01 97 cc 06 00 6a 05 00 00 bb 05 e6 62 01 04 02 03 f2 38 45 e9 15
[1010]
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$
```
Still only one register is available and 1010 doesnt correspond to anything.

gedger, 2022-08-05:
Looks to me like it's working now as you had socket timeouts in the first post. You now need to find your registers. My Solis inverter starts at 2999 for both holding and input registers. 

I contacted Solis support and asked for the modbus register specification for my inverter model, it took 2 months but they did supply this document.

ishanfdo18098, 2022-08-05:
2999 ?
All this time I was searching from 30000 to 39999 which was the range in the example.py 🤣
I will do a scan around that numbers and see. 

Thank you very much for your help

ishanfdo18098, 2022-08-05:
> 2999 for both holding and input registers.

[image]
Thank  you a lot. You saved my day

gedger, 2022-08-05:
Try my inverter settings they may well work for you.

```
	print("Date/Time :", modbus.read_holding_registers(register_addr=2999, quantity=6))
	print("Failsafe :", modbus.read_holding_registers(register_addr=3152, quantity=2))

	DC1V = modbus.read_input_register_formatted(register_addr=3021, quantity=1, scale=0.1)
	DC1A = modbus.read_input_register_formatted(register_addr=3022, quantity=1, scale=0.1)
	DC2V = modbus.read_input_register_formatted(register_addr=3023, quantity=1, scale=0.1)
	DC2A = modbus.read_input_register_formatted(register_addr=3024, quantity=1, scale=0.1)
	DCPwr =	modbus.read_input_register_formatted(register_addr=3006, quantity=2, scale=0.001)

	DC1Pwr = DC1V * DC1A / 1000
	DC2Pwr = DC2V * DC2A / 1000
	ACActPwr = modbus.read_input_register_formatted(register_addr=3004, quantity=2, scale=0.001)
	ACAppPwr = modbus.read_input_register_formatted(register_addr=3057, quantity=2, scale=0.001)

	print("\nAC Apparent Power: {:.2f}".format(ACAppPwr) , "kWh")
	print("AC Active Power: {:.2f}".format(ACActPwr) , "kWh")
	print("AC V: ", modbus.read_input_register_formatted(register_addr=3035, quantity=1, scale=0.1), "V")
	print("AC A: ", modbus.read_input_register_formatted(register_addr=3038, quantity=1, scale=0.1), "A")
	print("AC Freq: ", modbus.read_input_register_formatted(register_addr=3042, quantity=1, scale=0.01), "Hz")

	print("\nCalculated DC Power: {:.2f}".format(DC1Pwr + DC2Pwr), "kW")
	print("Current DC Power: {:.2f}".format(DCPwr), "kW")

	print("DC Power 1: {:.2f}".format(DC1Pwr), "kW")
	print("DC Power 2: {:.2f}".format(DC2Pwr), "kW")
	print("Efficiency: {:.1f}".format(ACActPwr / DCPwr * 100), "%")
	print("DC V 1: {:.1f}".format(DC1V), "V")
	print("DC A 1: ", DC1A, "A")
	print("DC V 2: {:.1f}".format(DC2V), "V")
	print("DC A 2: ", DC2A, "A")

	print("\nTemp: {:.1f}".format(modbus.read_input_register_formatted(register_addr=3041, quantity=1, scale=0.1)), "C")

	""" Note: Many of the yield figures are 32 bits but not sure which register, back one or forward one?? """
	print("\nPrev Month Yield: ", modbus.read_input_register_formatted(register_addr=3012, quantity=2), "kWh")
	print("This Month Yield: ", modbus.read_input_register_formatted(register_addr=3010, quantity=2), "kWh")
	print("Total Yield: ", modbus.read_input_register_formatted(register_addr=3008, quantity=2, scale=0.001), "MWh")
	print("Annual Yield: ", modbus.read_input_register_formatted(register_addr=3016, quantity=2, scale=0.001), "MWh")

	print("\nYear: ", modbus.read_input_register_formatted(register_addr=3072, quantity=1))
	print("Month: ", modbus.read_input_register_formatted(register_addr=3073, quantity=1))
	print("Day: ", modbus.read_input_register_formatted(register_addr=3074, quantity=1))
	print("Hour: ", modbus.read_input_register_formatted(register_addr=3075, quantity=1))
	print("Minute: ", modbus.read_input_register_formatted(register_addr=3076, quantity=1))
	print("Secs: ", modbus.read_input_register_formatted(register_addr=3077, quantity=1))

```

ishanfdo18098, 2022-08-05:
```
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$ python3 client_example.py 
Date/Time : [22, 8, 5, 13, 40, 7]
Failsafe : [1, 0]

AC Apparent Power: 0.49 kWh
AC Active Power: 0.49 kWh
AC V:  237.9 V
AC A:  2.1 A
AC Freq:  50.050000000000004 Hz

Calculated DC Power: 0.56 kW
Current DC Power: 0.56 kW
DC Power 1: 0.56 kW
DC Power 2: 0.00 kW
Efficiency: 88.3 %
DC V 1: 231.3 V
DC A 1:  2.4000000000000004 A
DC V 2: 3.0 V
DC A 2:  0.1 A

Temp: 39.0 C

Prev Month Yield:  51 kWh
This Month Yield:  30 kWh
Total Yield:  0.082 MWh
Annual Yield:  0.082 MWh

Year:  22
Month:  8
Day:  5
Hour:  13
Minute:  40
Secs:  9
ishanfdo@Dell-Inspiron:/mnt/c/Users/IshanFernando/Desktop/tmp/pysolarmanv5/examples$
```

Even better 🥳
Thanks a lot

ishanfdo18098, 2022-08-05:
> but they did supply this document.

can you send me the document ?

gedger, 2022-08-05:
There's several versions floating around, every inverter seems slightly different although the core seems common. 

[RS485_MODBUS Communication Protocol_RHI1P(5-10)K-HVES-5G-US_INV-3000ID EPM-36000ID 08.12.21.pdf](https://github.com/jmccrohan/pysolarmanv5/files/9268246/RS485_MODBUS.Communication.Protocol_RHI1P.5-10.K-HVES-5G-US_INV-3000ID.EPM-36000ID.08.12.21.pdf)

closed 2022-08-08 by ishanfdo18098