jmccrohan/pysolarmanv5 #17: Seme feedback

mzanetti, 2022-08-09:
While reading through this in an attempt to write something similar myself (thanks, this is quite helpful!) I've noticed this line:

https://github.com/jmccrohan/pysolarmanv5/blob/main/pysolarmanv5/pysolarmanv5.py#L101

To me this seems to be a request/response code, actually it seems to be 2 fields:
Setting the first one to a random number (or a counter), will cause the inverter to reply with the same number being set in the same field of response. This is useful to match a response to a request we've sent.

The second field seems to be a counter which counts up for every packet the inverter sends through the network. My assumption currently is that it's used for de-duplication. While the inverter doesn't seem to care what's in there, I suspect it may be good practice to set this to some unique value for every packet.

jmccrohan, 2022-08-09:
Hi @mzanetti,

Thanks for raising this. I guess I had never sat and watched it roll over before, so I never noticed that `serial` was effectively two single byte fields!

I'll look to split into two fields and use the first byte to an additional verification on the response.

Out of interest, are you intending to implement Solarman V5 in a different language, or are you trying to write a mobile app etc?

closed 2022-08-16 by jmccrohan

mzanetti, 2022-08-23:
> Out of interest, are you intending to implement Solarman V5 in a different language, or are you trying to write a mobile app etc?

Hey, I'm writing a plugin for nymea (so using C++/Qt). It's already working reasonably well but not 100% finished yet. Still need to do some cleanups: https://github.com/nymea/nymea-plugins/pull/591

commit edeee309 (2022-08-16): Implement sequence number validation

Many thanks to Michael Zanetti (@mzanetti) for highlighting this.

Github: Closes #17