jmccrohan/pysolarmanv5 #18: Sofar HYD 5000/6000-ES: V5 frame does not contain a valid Modbus RTU frame

Matthiasfranck, 2022-08-11:
Getting this error for my Sofar HYD 5000-ES inverter:

```
from pysolarmanv5 import PySolarmanV5, V5FrameError
import umodbus.exceptions
modbus = PySolarmanV5("192.168.65.240", 2356937823, port=8899, mb_slave_id=1, verbose=0)

retval = modbus.read_holding_registers(0x0210,4)
print(retval)
```

```
Traceback (most recent call last):
  File "solarman.py", line 5, in <module>
    retval = modbus.read_holding_registers(0x0210,4)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 370, in read_holding_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 266, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 254, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 222, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame

```

It seems like the RTU frame has a length of 2.

Any clue why this is happening? Is my inverter of a new type, which uses different messages?

Some information about the inverter:
[image]

jmccrohan, 2022-08-12:
@Matthiasfranck Can you enable verbose mode and post the SEND and RECV frames?

Matthiasfranck, 2022-08-12:
@jmccrohan Thanks for your reply

```
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 00 00 5f 04 7c 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 10 00 04 44 74 ab 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 10 00 10 15 00 a3 5f 04 7c 8c 02 01 17 bf 02 00 e0 27 00 00 c9 65 f3 62 05 00 ad 15
Traceback (most recent call last):
  File "solarman2.py", line 5, in <module>
    retval = modbus.read_holding_registers(0x0210,4)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 370, in read_holding_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 266, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 254, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 222, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame
```

jmccrohan, 2022-08-13:
@Matthiasfranck

You might need to try to sniff the RS-485 port to figure out what is going on. I know that other Sofar devices such as the ME-3000SP seem to have a Solarman data logging stick, but a separate RS-485 port, and it didn't seem to be possible to get pysolarmanv5 working on that device.

Looking at the photo of the HYD-6000 here, this could also be the case.
https://github.com/AndyWhittaker/HYD6000

Regards,
Jon

Matthiasfranck, 2022-08-24:
I have a workaround for now:

https://github.com/StephanJoubert/home_assistant_solarman/issues/80#issuecomment-1225656976

closed 2024-05-02 by jmccrohan