jmccrohan/pysolarmanv5 #20: logger unresponsive multiple times during the day

fnurglewitz, 2022-09-16:
Hi, I've been using this library for some months and everything was working perfectly until a couple weeks ago when I started noticing that the logger (LSW3_15_FFFF_1.0.57) is becoming unresponsive multiple times per day

I was wondering if you had any similar issue, I've also tried to stop polling the logger (I usually poll it every 2 minutes) and left a ping running and it still went down at some point

Not sure if I fried it by polling too often (but 2 minutes should not be an issue imo) or something else is happening

jmccrohan, 2022-09-18:
I have occasionally seen my logger (DLS-W, running MW_08_512_0501_1.82) become unresponsive and restart of its own accord. Nothing of the scale you're describing though.

Is your WiFi signal quality sufficient?

fnurglewitz, 2022-09-18:
Usually it's around 60%
Yesterday I tried to do a factory reset on the logger and today it happened "only" 3 times, not sure why though

fnurglewitz, 2022-09-21:
update: the factory reset actually helped, now the occurrence happens 1 or 2 times per day instead of 10+

still a mistery, closing the issue since it's not useful

closed 2022-09-21 by fnurglewitz