jmccrohan/pysolarmanv5 #21: Problem installing on RPi Zero W

usbrit, 2022-09-20:
Hi,
I am having issues install on RPi zero w with Python 3.10 installed.
Here's the error:
```
pizero@xxxxx:~ $  pip install git+https://github.com/jmccrohan/pysolarmanv5.git
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/jmccrohan/pysolarmanv5.git
  Cloning https://github.com/jmccrohan/pysolarmanv5.git to /tmp/pip-req-build-mUk6nk
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-mUk6nk/setup.py", line 3, in <module>
        with open("README.md", "r", encoding="utf-8") as fh:
    TypeError: 'encoding' is an invalid keyword argument for this function

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-mUk6nk/
```
Any help would be appreciated.
Thanks,
David

jmccrohan, 2022-09-20:
Are you using Python 3.8+ as per the minimum requirements?

Can you show the output of `pip debug`?

usbrit, 2022-09-21:
Thanks, re checked and although I thought I had upgraded, was still at 3.7
Sorry to trouble you

usbrit, 2022-09-21:
OK, so install seems to work, but I'm expecting to find a pysolarmanv5 directory in root so that solismod can the dictionary from it. There isn't one, and therefore solismod doesn't appear to communicate with the inverter.
This is what the install and directory list looks like
[image]
Solismod is receiving messages from mqtt, but doesn't change anything on the inverter
[image]

usbrit, 2022-09-23:
Figured it out, got some registers working, so connection is good!

jmccrohan, 2022-09-24:
Glad to hear you got it working.

Will close this issue.

closed 2022-09-24 by jmccrohan