jmccrohan/pysolarmanv5 #24: Single register write failed with Deye SUN300G3-EU-230

seepv, 2023-01-28:
The register doesn’t get written with this command:
`modbus.write_holding_register(register_addr=40, value=10)`
Result:
`pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame`

But it’s possible to read register_addr=40:
`modbus.read_holding_registers(register_addr=40, quantity=1)`
Result:
`[100]`

Is this a Bug or does anyone have an idea ?

seepv, 2023-01-28:
The Deye SUN300G3-EU-230 needs function code 16 instead of 6 for writing,
but the result is this Error:
`TypeError: object of type 'int' has no len()`

With the command:
`modbus.write_multiple_holding_registers(register_addr=40, values=10)`
Result:
```
Traceback (most recent call last):
  File "/data/deye/client_example-w.py", line 21, in <module>
    main()
  File "/data/deye/client_example-w.py", line 18, in main
    print(modbus.write_multiple_holding_registers(register_addr=40, values=10))
  File "/usr/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 463, in write_multiple_holding_registers
    mb_request_frame = rtu.write_multiple_registers(
  File "/usr/lib/python3.8/site-packages/umodbus/client/serial/rtu.py", line 169, in write_multiple_registers
    function.values = values
  File "/usr/lib/python3.8/site-packages/umodbus/functions.py", line 1510, in values
    if not (1 <= len(values) <= 0x7B0):
TypeError: object of type 'int' has no len()
```

Is this TypeError an error like #2 
or something different ?

seepv, 2023-01-28:
Sorry - my False-Typo - no square brackets.

With the Command:
`modbus.write_multiple_holding_registers(register_addr=40, values=[10])`
Result:
`1`

And Register-Read:
`modbus.read_holding_registers(register_addr=40, quantity=1)`
Result:
`[10]`

closed 2023-01-28 by seepv

gianricod, 2023-02-06:
hi @seepv 
while this is working on my 2000g3 the actual power is not reduced at all
is the power being reduced for you?

schwatter, 2023-03-23:
Back to the future.

I have some strange effect. When i write

`modbus.write_multiple_holding_registers(register_addr=40, values=[90])`

Nothing happen. After that i write:

`modbus.write_multiple_holding_registers(register_addr=40, values=[80])`

But 90 % appaers. So always the last one, not the actual one i set....
So i have to write twice to get my acctual value.

Any hint?

schwatter, 2023-03-23:
Oh sorry,

my mistake, i was lost in my own code....
I did a mistake in my sequence.