jmccrohan/pysolarmanv5 #41: question :  write holding register

chinezbrun, 2023-08-30:
not found the right sintax to write a register via pysolarman5 to Deye inverter

example 
print(modbus.read_holding_registers(register_addr=99,quantity=1))
>> 5760 

modbus.write_holding_register(register_addr=99,value=5750)

give me the errorr
"raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame"

githubDante, 2023-08-30:
Deye only supports writing to multiple registers (function code 0x10). Use:
```
modbus.write_multiple_holding_registers(99, [5750])
```
and should be OK.

chinezbrun, 2023-08-31:
thanks a lot . all good with the syntax provided

closed 2023-08-31 by chinezbrun