jmccrohan/pysolarmanv5 #44: Thread Error (concurrency), Socket closed or lost

jlopez77, 2023-09-27:
Hi!

I'm using your integration to write an API for my Deye Inverter but I'm facing errors that force me to restart the API whenever the connection is not used for 3 to 10 minutes.
I'm using the auto_reconnect option and **sometimes** it works but eventually, when a read or write operation is performed, the following error is thrown and cannot be recovered.

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 299, in _data_receiver
    data = self.sock.recv(1024)
ConnectionResetError: [Errno 104] Connection reset by peer

githubDante, 2023-09-27:
This is definitely a discrepancy between the sync and async versions. Connection reset is handled in the [latter](https://github.com/jmccrohan/pysolarmanv5/blob/c0221a7c13607a61e4aa3e8131b0bb16786cbb2b/pysolarmanv5/pysolarmanv5_async.py#L135C1-L138C32). A try/except block assigning empty bytes to the `data` variable on reset should fix it.

jlopez77, 2023-09-27:
That's what I ended up doing :)

githubDante, 2023-09-27:
Thanks for the report. If @jmccrohan  knows how to add you to the co-authors that would be great :)

jmccrohan, 2023-09-27:
Thanks @jlopez77 and @githubDante. I've pushed b87f4a0 now which resolves this and #45.

Btw, you can add  `Co-authored-by:` to commit messages to credit additional developers.

closed 2023-09-27 by jmccrohan