jmccrohan/pysolarmanv5 #49: Solis S5GR3PXX, tested on Solis S5GR3P20K

DamithaKD, 2023-12-29:
hi guys 

not sure where to post this This is the register map for the Solis S5GR3P20K 
hope this helps some one  

note :  i found few more values and ill update as i find them 

Register: 03004         Value: 00000 (0x0000) AC Power >>
Register: 03005         Value: 00620 (0x026c) Ac Power * 1000
Register: 03006         Value: 00000 (0x0000) DC Power >>
Register: 03007         Value: 00754 (0x02f2) DC Power  * 1000 KW
Register: 03008         Value: 00000 (0x0000) total kwh >>
Register: 03009         Value: 00224 (0x00e0) total kwh * 1
Register: 03010         Value: 00000 (0x0000) Monthly KWH >>
Register: 03011         Value: 00224 (0x00e0) Monthly KWH * 1 
Register: 03012         Value: 00000 (0x0000) Last Monthly KWH >>
Register: 03013         Value: 00000 (0x0000) Last Monthly KWH * 1 (Not sure )
Register: 03014         Value: 00246 (0x00f6) Daly Kwh  * 10
Register: 03015         Value: 00201 (0x00c9) yestaday KWh * 10 
Register: 03016         Value: 00000 (0x0000) Yearly KWH >>
Register: 03017         Value: 00224 (0x00e0) YEARLY kwh * 1 

Register: 03021         Value: 03740 (0x0e9c) DC1 V * 10
Register: 03022         Value: 00021 (0x0015) DC1 A * 10
Register: 03023         Value: 00026 (0x001a) DC2 V * 10
Register: 03024         Value: 00000 (0x0000) DC2 A * 10

Register: 03033         Value: 02332 (0x091c) VR  * 10
Register: 03034         Value: 02324 (0x0914) VY * 10
Register: 03035         Value: 02325 (0x0915) VB * 10
Register: 03036         Value: 00009 (0x0009) IR * 10 
Register: 03037         Value: 00010 (0x000a) IY * 10
Register: 03038         Value: 00010 (0x000a) IB * 10
Register: 03039         Value: 00000 (0x0000)
Register: 03040         Value: 00000 (0x0000) Working Mode 
Register: 03041         Value: 00396 (0x018c) Temp * 10
Register: 03042         Value: 05003 (0x138b) Frequency  * 100
Register: 03043         Value: 00003 (0x0003) Grid Mode 

Register: 03072         Value: 00023 (0x0017) Year 
Register: 03073         Value: 00012 (0x000c) Month 
Register: 03074         Value: 00021 (0x0015) Date 
Register: 03075         Value: 00016 (0x0010) Hour 
Register: 03076         Value: 00057 (0x0039) MIn 
Register: 03077         Value: 00004 (0x0004) Sec

DamithaKD, 2024-01-02:
Register: 02999         Value: 01291 (0x050b) 	Product Version   : Not Sure 
Register: 03000         Value: 00244 (0x00f4) 	DSP Version  : Not Sure 
Register: 03001         Value: 00036 (0x0024) 	LCD Version : Not Sure 
Register: 03002         Value: 00001 (0x0001)
Register: 03003         Value: 00001 (0x0001)
Register: 03004         Value: 00000 (0x0000) 	AC Power (MSB)
Register: 03005         Value: 00620 (0x026c) 	AC Power * 1000 KW 	( Signed  INT 32 bit ) 
Register: 03006         Value: 00000 (0x0000) 	DC Power (MSB)
Register: 03007         Value: 00754 (0x02f2) 	DC Power * 1000 KW 	( Signed  INT 32 bit) 
Register: 03008         Value: 00000 (0x0000) 	Total kWh (MSB)
Register: 03009         Value: 00224 (0x00e0) 	Total kWh * 1 		(INT  32 bit)
Register: 03010         Value: 00000 (0x0000) 	Monhtly kWh (MSB)		
Register: 03011         Value: 00224 (0x00e0) 	Monthly kWh * 1 	(INT  32 bit)	
Register: 03012         Value: 00000 (0x0000) 	Last Month kWh (MSB)
Register: 03013         Value: 00000 (0x0000) 	Last Month KWh * 1  (INT  32 bit)	
Register: 03014         Value: 00246 (0x00f6) 	Daly kWh  * 10
Register: 03015         Value: 00201 (0x00c9) 	Yestaday kWh * 10 
Register: 03016         Value: 00000 (0x0000) 	Yearly kWh (MSB)
Register: 03017         Value: 00224 (0x00e0) 	Yearly kWh * 1 		(INT  32 bit)
Register: 03018         Value: 00000 (0x0000)	Last Year kWh (MSB)
Register: 03019         Value: 00000 (0x0000) 	Last Year kWh * 1   (INT  32 bit)
Register: 03020         Value: 00036 (0x0024)
Register: 03021         Value: 03740 (0x0e9c) 	DC1 V * 10
Register: 03022         Value: 00021 (0x0015) 	DC1 A * 10
Register: 03023         Value: 00026 (0x001a) 	DC2 V * 10
Register: 03024         Value: 00000 (0x0000) 	DC2 A * 10
Register: 03025         Value: 00000 (0x0000)
Register: 03026         Value: 00000 (0x0000)
Register: 03027         Value: 00000 (0x0000)
Register: 03028         Value: 00000 (0x0000)
Register: 03029         Value: 00000 (0x0000)
Register: 03030         Value: 02979 (0x0ba3)
Register: 03031         Value: 05960 (0x1748)
Register: 03032         Value: 02978 (0x0ba2)
Register: 03033         Value: 02332 (0x091c) 	VR * 10
Register: 03034         Value: 02324 (0x0914) 	VY * 10
Register: 03035         Value: 02325 (0x0915) 	VB * 10
Register: 03036         Value: 00009 (0x0009) 	IR * 10 
Register: 03037         Value: 00010 (0x000a) 	IY * 10
Register: 03038         Value: 00010 (0x000a) 	IB * 10
Register: 03039         Value: 00000 (0x0000)
Register: 03040         Value: 00000 (0x0000) 
Register: 03041         Value: 00396 (0x018c) 	Temp * 10
Register: 03042         Value: 05003 (0x138b) 	Frequency  * 100
Register: 03043         Value: 00003 (0x0003) 	Grid Mode 
    
	This is Correct as Far As i can check and seems to match ( mostly ) 
    Value	 Text 	 									Tested
	
	0x0    : "Waiting State :"  						OK 
    0x1    : "Open Loop Operation :"
    0x2    : "Soft Start :"
    0x3    : "On Grid/Generating :" 					OK 
    0x1004 : "Grid OverVoltage :"         
    0x1010 : "Grid UnderVoltage :"        
    0x1012 : "Grid OverFrequency :"         
    0x1013 : "Grid UnderFrequency :"          
    0x1014 : "Grid Imp too large :"
    0x1015 : "No Grid :"      							OK   
    0x1016 : "Grid Imbalance :"         
    0x1017 : "Grid Freq Jitter :"         
    0x1018 : "Grid Overcurrent :"         
    0x1019 : "Grid Tracking Fault :"         
    0x1020 : "DC OverVoltage :"         
    0x1021 : "DC Bus Overvoltage :"
    0x1022 : "DC Bus Uneven Voltage :"       
    0x1024 : "DC Bus Uneven Voltage2 :"
    0x1025 : "DC A path OverCurrent :"
    0x1026 : "DC B path OverCurrent :"        
    0x1027 : "DC Input Disturbance :"       
[... 52 lines cut ...]
Register: 03068         Value: 00000 (0x0000)
Register: 03069         Value: 00000 (0x0000)
Register: 03070         Value: 00000 (0x0000)
Register: 03071         Value: 00001 (0x0001) Working Mode
                
    This is Correct as Far As i can check and seems to match ( mostly ) 
	Value	 Text 	 									Tested
	
	0x01   : "Normal Operation :" 						OK 
    0x02   : "Initial Standby :"
    0x04   : "Control Shutdown :"
    0x08   : "Downtime:" 								OK 
    0x10   : "Standby:" 								OK 
    0x20   : "Derating Operation:"   
    0x40   : "Limit Operation:"
    0x80   : "Bypass Overload:"
				
Register: 03072         Value: 00023 (0x0017) 	Year 
Register: 03073         Value: 00012 (0x000c) 	Month 
Register: 03074         Value: 00021 (0x0015) 	Date 
Register: 03075         Value: 00016 (0x0010) 	Hour 
Register: 03076         Value: 00057 (0x0039) 	Min 
Register: 03077         Value: 00004 (0x0004) 	Sec

Note : seems to be mostly 00 form here on

jmccrohan, 2024-05-02:
Not specifically related to pysolarmanv5. Closing.

closed 2024-05-02 by jmccrohan