jmccrohan/pysolarmanv5 #50: Data once every second

teaalltr, 2024-01-18:
Is it possible to get data at this rate? What's the max rate you achieved?
In case, what are the optimizations/tricks to apply to the basic python example script?

githubDante, 2024-01-20:
It should be possible, but still it depends on the inverter and the connection. Some benchmarks below:

``` python
>>> logger_bench(logger, 1)
10 reads of 1 registers in 1.793 seconds: min [169.56ms], max [182.37ms], avg[179.191ms]
>>> logger_bench(logger, 1)
10 reads of 1 registers in 2.044 seconds: min [169.62ms], max [340.2ms], avg[204.341ms]
>>> logger_bench(logger, 1)
10 reads of 1 registers in 1.803 seconds: min [170.29ms], max [195.68ms], avg[180.186ms]
>>> logger_bench(logger, 10)
10 reads of 10 registers in 2.267 seconds: min [189.83ms], max [369.92ms], avg[226.683ms]
>>> logger_bench(logger, 10)
10 reads of 10 registers in 1.921 seconds: min [189.44ms], max [201.47ms], avg[192.006ms]
>>> logger_bench(logger, 10)
10 reads of 10 registers in 2.014 seconds: min [198.44ms], max [206.77ms], avg[201.374ms]
>>> logger_bench(logger, 100)
10 reads of 100 registers in 3.934 seconds: min [389.59ms], max [400.59ms], avg[393.375ms]
>>> logger_bench(logger, 100)
10 reads of 100 registers in 4.151 seconds: min [389.79ms], max [493.44ms], avg[414.993ms]
>>> logger_bench(logger, 100)
10 reads of 100 registers in 4.21 seconds: min [388.73ms], max [555.01ms], avg[420.957ms]
>>> 

```

and logger bench method:

``` python
def logger_bench(logger: PySolarmanV5, reg_len: int):
    t = []
    fs = time()
    for i in range(10):
        s = perf_counter()
        _ = logger.read_holding_registers(60, reg_len)
        t.append(round(1000 * (perf_counter() - s), 2))
    
    print("10 reads of {} registers in {} seconds: min [{}ms], max [{}ms], avg[{}ms]".format(reg_len, round(time() - fs, 3), min(t), max(t), round(sum(t)/len(t), 3)))
```

paaspaas00, 2024-02-21:
@githubDante thank you!

closed 2024-05-02 by jmccrohan