jmccrohan/pysolarmanv5 #51: HYD5000 HP: Same call, sometimes getting V5FrameError with concurrent clients

virtualdj, 2024-01-20:
Hi, I'm using your wonderful library with a Sofar/ZCS HYD5000 HP with a Wi-Fi USB stick with firmware LSW3_15_270A_1.68 and serial number starting with 29.

My Python test script is very basic and based on your example:
```python
from pysolarmanv5 import PySolarmanV5

def main():
    modbus = PySolarmanV5(
        "192.168.0.55", 2900000000, port=8899, mb_slave_id=1, verbose=True, socket_timeout=15
    )
    print(modbus.read_holding_registers(register_addr=0x604, quantity=1))
    modbus.disconnect()

if __name__ == "__main__":
    main()
```
When running that code I sometimes get the error `V5FrameError: V5 frame contains invalid sequence number`:
```text
DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=1028, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.7', 59795), raddr=('192.168.0.55', 8899)>
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 85 00 d8 5e 18 ad 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 06 04 00 01 c5 43 05 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 43 00 10 15 a8 8f d8 5e 18 ad 02 01 10 f7 0a 00 63 08 00 00 56 fe a0 65 01 03 30 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 24 00 00 00 18 00 00 e4 c1 8b 15
Traceback (most recent call last):
  File "c:\pytest.py", line 45, in <module>
    main()
  File "c:\pytest.py", line 23, in main
    print(modbus.read_holding_registers(register_addr=0x604, quantity=1))
  File "...\Python310\site-packages\pysolarmanv5\pysolarmanv5.py", line 512, in read_holding_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "...\Python310\site-packages\pysolarmanv5\pysolarmanv5.py", line 389, in _get_modbus_response
site-packages\pysolarmanv5\pysolarmanv5.py", line 377, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "...\Python310\site-packages\pysolarmanv5\pysolarmanv5.py", line 249, in _v5_frame_decoder
    raise V5FrameError("V5 frame contains invalid sequence number")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame contains invalid sequence number
```
But when running the same code again, **without any change**, one or two times I eventually get the data back:
```text
DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=924, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.7', 59798), raddr=('192.168.0.55', 8899)>
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 01 00 d8 5e 18 ad 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 06 04 00 01 c5 43 81 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 01 9b d8 5e 18 ad 02 01 13 f7 0a 00 66 08 00 00 56 fe a0 65 01 03 02 01 fa 39 97 80 15
[506]
```
I'm running this on a PC, while I have another Home Assistant instance on another PC that queries the same inverter with your library once a minute.

Is the **V5 frame contains invalid sequence number** error related to the concurrency of the different clients? If yes, is it by design, provided that they are using a different IP? So, is this a limitation of the stick?

I also noticed, with the same inverter and without any concurrency, that if I ask a **quantity** greater than 60 I always get a `umodbus.client.serial.redundancy_check.CRCError: CRC validation failed` exception.
As I'm able to workaround by doing multiple calls, I'm just asking if this is something that happens to your devices, too.

githubDante, 2024-01-20:
> Is the V5 frame contains invalid sequence number error related to the concurrency of the different clients? If yes, is it by design, provided that they are using a different IP? So, is this a limitation of the stick?

Yes, it is a concurrency problem caused by the limitations of the stick. The latter broadcasts the repspones to all connected clients. Try to update directly from Github as the main branch already has patches in this regard.

> I also noticed, with the same inverter and without any concurrency, that if I ask a quantity greater than 60 I always get a umodbus.client.serial.redundancy_check.CRCError: CRC validation failed exception.

No expirience with Sofar, but this looks like a limitation of the inverter. The protocol allows up to 125 registers per request.

virtualdj, 2024-01-20:
Thanks for the answer; I'll close the issue as it was meant to be only a question.

closed 2024-01-20 by virtualdj