jmccrohan/pysolarmanv5 #55: Issue with writing to register on Sofar HYD 10KTL-3PH Inverter Using pysolarmanv5

pmenclik, 2024-05-02:
Hello,

I am working with a Sofar Solar HYD 10KTL-3PH inverter, interfacing using a Solarman logger version LSW3_15_270A_1.68. I have successfully managed to read data from the inverter using the pysolarmanv5 library, but I encounter errors when attempting to write to the inverter's registers.

**Issue Description:**
Reading from the inverter's registers works without any issues; however, when I attempt to write to the registers to set new Reflux values, the process fails with a Modbus frame error indicating that the frame does not contain a valid Modbus RTU frame.

**Technical Details:**
Library Used: pysolarmanv5
Functionality Affected: Writing to registers
Error Message: "V5 frame does not contain a valid Modbus RTU frame"
Communication: Modbus TCP/IP
Inverter Type: Sofar Solar HYD 10KTL-3PH
Logger Version: LSW3_15_270A_1.68

Script Snippet and Full Hexadecimal Data:
Here is the relevant part of my script attempting to write to a register, along with the hexadecimal data logged by the script:

**Code:**
from pysolarmanv5 import PySolarmanV5

def main():
    modbus = PySolarmanV5(
        "192.168.x.x", 234xxxxxx7, port=8899, mb_slave_id=1, verbose=True
    )
    print("Current setting:")
    reflux = modbus.read_holding_register_formatted(register_addr=0x1023, quantity=1)
    print(f"Register: {0x1023:5}\tValue(0-reflux on, 1-no reflux, 2-limit): {reflux}")

    new_reflux_value = 2
    print(f"Attempting to write new value: {new_reflux_value} to register {0x1023:5}")
    write_response = modbus.write_holding_register(0x1023, new_reflux_value)
    print("Write response:", write_response)

    print("Verifying change...")
    reflux = modbus.read_holding_register_formatted(register_addr=0x1023, quantity=1)
    print(f"Updated Register: {0x1023:5}\tValue: {reflux}")

    modbus.disconnect()

if __name__ == "__main__":
    main()

**Verbose Log Output:**

DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=3, family=2, type=1, proto=6, laddr=('192.168.117.199', 50580), raddr=('192.168.91.31', 8899)>
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 46 00 c3 94 e1 8b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 10 23 00 01 71 00 20 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 46 2b c3 94 e1 8b 02 01 52 3e 8c 02 90 0b 00 00 70 3c a7 63 01 03 02 00 01 79 84 e4 15
Register:  4131	Value(0-reflux on, 1-no reflux, 2-limit):  1
Writing new value:  2
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 47 00 c3 94 e1 8b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 06 10 23 00 02 fd 01 b2 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 10 00 10 15 47 2c c3 94 e1 8b 02 01 54 3e 8c 02 92 0b 00 00 70 3c a7 63 05 00 e6 15
Traceback (most recent call last):
  File "pysolarmanv5_write_test.py", line 36, in <module>
    main()
  File "pysolarmanv5_write_test.py", line 26, in main
    print(modbus.write_holding_register(register, new_reflux_value))
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame

**Request for Assistance:**
Has anyone experienced similar issues with writing to Sofar Solar inverters using the pysolarmanv5 library, or does anyone have insights into resolving this frame error? Any suggestions or recommendations would be greatly appreciated.

Thank you in advance for your support and assistance!

jmccrohan, 2024-05-02:
Hi @pmenclik,

Based on what you've supplied there, pysolarmanv5 seems to be working correctly. I've highlighted the Modbus RTU frame between `**` below.

This is the read request and response from your output above:
```
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 46 00 c3 94 e1 8b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 **01 03 10 23 00 01 71 00** 20 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 15 00 10 15 46 2b c3 94 e1 8b 02 01 52 3e 8c 02 90 0b 00 00 70 3c a7 63 **01 03 02 00 01 79 84** e4 15
```

While this is the write request and response:
```
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 47 00 c3 94 e1 8b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 **01 06 10 23 00 02 fd 01** b2 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 10 00 10 15 47 2c c3 94 e1 8b 02 01 54 3e 8c 02 92 0b 00 00 70 3c a7 63 **05 00** e6 15
```
`05 00` is not a valid RTU frame.

I suggest you talk to @Matthiasfranck. They reported the same thing with a Sofar inverter in #18.

Regards,
Jon

jmccrohan, 2024-05-02:
> Hi, thx very much for answer. In meantime I found working solution if anybody might be concerned.
> 
> When writing to registers of Sofar you have to use modbus.write_multiple_holding_registers(register, reflux), where register is 4131 (decimal) and reflux is list of two values [2,71] 2 - feedin mode, 71 = 7.1KW limit power. If I want to limit feedin, I just write [2,0] which sets feedin-limit power. That works for me.
> 
> Thx for the library, it solves my problem!

Thanks for getting back and for sharing the quirk of the Sofar Modbus implementation.

FYI @Matthiasfranck @d-eggert

closed 2024-05-02 by jmccrohan