jmccrohan/pysolarmanv5 #56: SolarMan app (LSW-3 logger) hides parameters. Could they be obtained using pysolarmanv5?

solar4ever, 2024-05-17:
Hi all,

First of, please accept my apologies for posting/asking this here, but there doesn't seem to be any other place to seek for help, regarding SolarMan, anymore.

About two weeks ago I had my solar panel system installed, which uses an Azzurrro Inverter (Model No: **AZZURRO 1PH 5000TLM-V3**) and a SolarMan Wifi Stick **Logger LSW-3** (Module Version No: **LSW3_15_270A_1.68**, Extended System Version: **V1.1.00.10**). My installer set the logger up and gave me authorization on the SolarMan Smart app. Until recently, everything worked fine and I was able to see all the parameters available in both the mobile app and website (using _globalhome.solarmanpv.com_), as well as the Excel spreadsheets that I exported using their website. 

About a week ago, various **parameters got dissappeared** from both the app and the website, as well as the exported spreadsheets, while more params keep getting removed up to this date. It all started with **Single Plate Ambient Temperature** and **Radiator Temperature 1** parameters - which are the most important to me - but more of them have been removed in the following days, including _leak current_, _hardware version_, _string_ information, _inverter working status_, _alert code_s, _control_ information and more. 

Contacting Solarman's customer service (which was not very helpful - broken English, not paying attention to email's content and not answering my questions, as well as it felt like talking to scammers), they said that there is nothing wrong with their platform or the Sofar protocol, which is used by Azzurro, but instead, those parameters were masked by the seller of my inverter. My installer, though, **claims that he did not block** any of these parameters. The logger was set up two weeks ago and everything was running ok, and nobody physically touched the inverter since then. So, my questions are:

**1.** Could the installer remotely lock these parameters though SolarMan Businness app or Solarman website at _pro.solarmanpv.com_? Could the installer do that through Azzurro's Portal or some other platform using the inverter's S/N?

**2.** If the installer did hide these paramters from me, did he do that on the inverter's side (on the inverter itslef or on Azzurro's web server) or the logger's side (on the adapter itself or SolarMan's web server)? So, resetting or even buying a new logger (without giving away the new S/N), should I be able to see these parameters again?

**3.** Does **pysolarmanv5** support my logger (see above for model info)? If yes, using pysolarmanv5, would I be able to obtain  "Single Plate Ambient Temperature" and "Radiator Temperature 1" parameters, as well as every other parameter that it is now gone from SolarMan Smart app and website, and how (what is the register number for each)? Could pysolarmanv5 likely cause any damage/issues to the logger, if using it?

**4.** Has anyone else ever been in a similar situation, using similar setup to mine (Azzurro and Solarman logger), and could anyone confirm whether or not these parameters appear in their plant?

Thanks, and my apologies once again.

githubDante, 2024-05-21:
Hi,

Straight to your questions:

1. Highly unlikely. It's possible probably through a firmware update, but these parameters are also used by the inverter, so why would they whant to hide them.

2. See 1, the parameters are probably hidden in the web interface. 

3. It should be supported. There is a long discussion regarding these inverters [here](https://github.com/StephanJoubert/home_assistant_solarman/discussions/148) (mostly in Italian). In order to obtain the register values (and parse them) you need the Modbus documentation for the inverter. Try to request it directly from the manufacturer. 
And no, the logger/inverter will not be damaged if you read parameters from it.

4. Can confirm that these parameters are visible in the solarman platform (pro.solarmanpv.com) for Deye inverters.

solar4ever, 2024-05-21:
@githubDante First of all, thanks a lot for your response. It means a lot.

1. The reason I believe the installer had something to do with it is (a) because we had a discussion on the temperature subject, as the cover/box I installed around the inverter (after following his instructions on how to build it) proved to be a bit problematic, as the inverter would reach a **radiator temperature** of 85°C and above (and that is actually my main concern and the reason I need to watch it for now), and after an hour we hanged up, the temperature parameters were removed, (b) the response from solarman customer service (which was terrible so far) was that those _"parameters were masked by the seller of the inverter"._

2. I also thought that it might be the interface, but the solarman customer service said that **it is not an issue with their platform** **nor the sofar protocol** that Azzurro Inverter uses, but instead that the _"parameters were masked by the seller of the inverter_".

3. Yes. I have the scanner (see `pysolarmanv5` examples directory) running to find out the register values. Scanning holding registers was much faster and done already, but scanning for input registers takes much longer - plus, it seems that Sofar does not follow the rules of `range(30000, 39999)` for input registers and `range(40000, 49999)` for holding ones (as I found holding registers in many other ranges and certainly not between the above), meaning that I need to start scanning from value `1`.  I searched online but did not find the documentation for my model. Do you think they would provide such information? 

4. OK. So, if you can see those parameters on the business platform (pro.solarmanpv.com), even if it for Deye inverters, it means that **it is likely not an interface or firmware issue**. Azzurro **does** provide temperatures as well, and like I said above, everything worked fine until around a week ago (after that phone call). Since you are using the pro/business platform - note that I created an account myself, but did not let me to add the S/N of my logger cause it says it has already been added to a plant (by my installer). I wish there was a way to add it on the pro platform as well - so, **please**, tell me: 
(a) Does the pro platform allow you to change settings on the inverter (I guess it does. I saw a video that can let you set/read parameters - which uses the modbus protocol behind the scenes I guess)? 
(b) Does it let you to mask/lock parameters, such as temperature, so that it won't be visible to either the installer or the client (meaning that the change took place on the inverter and not their platform)? 
(c) Does it let you to mask/block parameters from being visible only to clients, but still be visible to the installer (meaning that the change only took place on their platform)?

githubDante, 2024-05-21:
1 & 2 It sounds bad. As the owner of the inverter you must have full access to data.

3. The example is only for reference and does not work for all inverters. You should know the address range and the function which to be used.
> Do you think they would provide such information? 

Probably they will. 

4. 
    (a) - No it is not allowed to change parameters. It needs authorization from the manufacturer. Deye support asked me for username and password :smiley:  in order to activate it, but I was working already on the `deye-controller` so I didn't bother with that.
    
    (b & c) - Just played with this. By creating a new user and granting him plant authorization that user can log in via globalhome.solarmanpv.com. It can see almost all parameters (including the temperatures) in `View Plant Only Mode` and only a simple Overview when the authorization is changed to `Ordinary Member`. There are missing sections on the right (Device, Authorization etc.). The temperatures can be seen from the Device section. So your suspicion is correct.
    
    This is how the authorization menu looks.

[image]

 Ask your so called `installer` for full access. It is your inverter after all.

solar4ever, 2024-05-22:
@githubDante Thank you once again for your response, but you got me confused at some points.

1. First, that's exactly how I think of it as well. I am the owner of the inverter and I should have full access. But, you know that you can't just install a solar system without going through goverrment authorities inspections etc, and have the installer update the inverter with the settings permitted for your country (frequencies etc). So, not sure how power they have as installers over that matter, but I should still ask for admin role as well, but he would probably deny it, and thus, I might end up buying a new logger myslef eventually and not letting him know of the S/N (what's your opinion on that?). When he first installed it, he argued that he should be admin and give the authorization to me; otherwise, my inverter warranty would be voided (cause I guess he wouldn't be able to see alerts that have to do with inverter faults etc. But it might be just an excuse).

2. You said that the pro platform does not allow changing parameters, unless you get authorization from the manufacturer. So, **it does** let you perform changes on the inverter afterall, right? And what kind of changes those might be, like the ones I mentioned? He might already have the authorization as an installer. They asked you for username and password of your Solarman account? That sounds weired.

3. Thank you for posting this screenshot. So, even in ordinary mode you could still see the temperatures. So, please let me know **if you can individually hide/mask parameters** from the client through the pro platform/business app. For example, I watched this video https://www.youtube.com/watch?v=IWet_Qd58kE that shows how to change settings in Solarman Business app - it does not show the temperature parameters, but I guess those might be there as well (at least, for inverters that support it, like my Azzurro and yours). Is "all pant authority role" above "admin", or not?

3. If you give "admin" access to the new user you created, do they have to be registered on pro/business platform I guess, in order for them to see everything? They wouldn't see any difference on the smart/globalhome platform, right?

Thank you.

githubDante, 2024-05-22:
Hi, 

1. In your setup data access is available on both platforms. The restrictions are controlled from the `pro`, so the installer still can see the alerts and all info regarding the inverter
> I might end up buying a new logger myslef eventually and not letting him know of the S/N (what's your opinion on that?). 

What will happen with your warranty in this case ?!? The so called `installer` will lose access completely.

2. Yes, after authorization from the manufacturer some settings can be changed, don't know which.

3. No, in Ordinary Mode only an overview is available, the temperatures remain hidden. As I said earlier there are missing sections in the menu. It looks like you are in such mode at the moment. 
The platform does not look advanced enough to allow hiding of specific parameters.

* Ordinary mode 
[image]

* All plant authority  
[image]

4. As an Admin the globalhome page is almost empty, so probably the `pro` should be used, but it needs a business account. It won't let you log in with the one from `globalhome`.

The sweet spot in your situation is `All plant authority role` or at least `View plant role`

As this issue is not related to pysolarman at all let's close it. When you have the modbus documentation you should be able to read the temperatures directly from the inverter.

solar4ever, 2024-05-22:
@githubDante Thank you once again for the repsonse and the details.

It looks like I am in `All plant authority` role or something. I can see all the menu items as in the screenshot you provided, besides the `Add a logger` option at the top (I am loging in through `globalhome`).

I emailed Sofar, but they did not response as yet. Regardless, I 've managed to find the temperature registers (i.e., `1048` and `1050`) and can now get the temperature values, which means that they are not masked/locked on the inverter, but whatever is happenning, it must be on solarman's side (or the logger's hardware maybe?). However, they keep saying there is nothing wrong with their platform, and now they ask me to contact Sofar (frankly, this must be the worst customer service I've ever encoutered). 

> The platform does not look advanced enough to allow hiding of specific parameters.

I was downloading the Excel spreadsheets every night since the logger was installed, and I can see the temperature columns, alarm codes, leak current and others that are now gone from spreadsheets as well - even for those dates before those parameters disappeared (i.e., if, today, I re-download Excel spreadsheets for the dates that those parameters were still available, those parameters columns will now be missing as well). That is why it looks like someone must have locked/blocked them from my account. If you are so sure that there is no such an option for hiding parameters (I guess you have seen the video I linked earlier?), then I don't know what to think anymore, as the customer service said that they did not change their platform, and the sofar protocol doesn't seem to have the register addresses for those parameters changed (I double checked with others online that have posted Excel spreadsheets for Sofar's modbus protocol in the previous years and the register addresses are still the same).

I agree. I will close it as soon as I get your response to this comment, in case you have something else to add to this conversation. You have been very helpful and I genuinely thank you for all the time and effort you put on trying to help me out (ragardless that we still haven't found out what is going on here). Thanks once again.

closed 2024-05-25 by solar4ever