jmccrohan/pysolarmanv5 #60: Sofar HYD 3600-ES "V5 frame does not contain a valid Modbus RTU frame"

ssimoe50, 2024-07-06:
Hello,

I'm facing an issue and difficulties with my recent installation.
 - Sofar HYD 3600-ES
 - Device information:
   - Device serial number: 239XXXXXXX
   - Firmware version: LSW3_15_270A_1.68
   - Connected through WiFi

Just to be sure I tried to scan and it was in fact detected this model:

```bash 
$ python3 scanSolarMan.py

{'ipaddress': '192.168.X.Y', 'mac': 'E8FDZZZZZZZZ', 'serial': '239XXXXXXX'}

```

But when I tried to run the basic example (with the correct IP and serial information of course I got this:
```bash
$ python3 basicSyncClientTest.py 

Traceback (most recent call last):
(...)
File "testsSolarman/basicSyncClientTest.py", line 17, in main
    print(modbus.read_input_registers(register_addr=33022, quantity=6))
(...)
raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame
```

Has anyone seem similar and/or has a solution or workaround for this in particular?

Thanks

jmccrohan, 2024-07-06:
Hi,

Can you enable debug logging please so that I can view the sent and receive
frames?

Cheers.

ssimoe50, 2024-07-06:
Yeah of course @jmccrohan !

Here they go:
```bash 
2024-07-06 13:13:28,954 - pysolarmanv5.pysolarmanv5 - DEBUG - Socket setup completed... <socket.socket fd=3, family=2, type=1, proto=6, laddr=('192.168.1.82', 52612), raddr=('192.168.1.119', 8899)>
2024-07-06 13:13:28,954 - pysolarmanv5.pysolarmanv5 - DEBUG - SENT: a5 17 00 10 45 3b 00 a2 72 f3 8e 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 80 fe 00 06 38 38 37 15
2024-07-06 13:13:40,610 - pysolarmanv5.pysolarmanv5 - DEBUG - RECD: a5 10 00 10 15 3b a1 a2 72 f3 8e 02 01 16 da 4b 00 8c 17 00 00 de 5a 3d 66 05 00 67 15

Traceback (most recent call last):
(...)
File "testsSolarman/basicSyncClientTest.py", line 17, in main
    print(modbus.read_input_registers(register_addr=33022, quantity=6))
(...)
raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame

```

githubDante, 2024-07-06:
This is a modbus exception code (0x05). Try with different address. Quick search for the inverter shows that the total buy/sell from/to grid should be in the range 0x021E (decimal 542) + 4 i.e the request should be:
```python
print(modbus.read_input_registers(register_addr=542, quantity=4))
```

ssimoe50, 2024-07-06:
Hi @githubDante 

> This is a modbus exception code (0x05). Try with different address. Quick search for the inverter shows that the total buy/sell from/to grid should be in the range 0x021E (decimal 542) + 4 i.e the request should be:
> 
> ```python
> print(modbus.read_input_registers(register_addr=542, quantity=4))
> ```

Based on your comment I attempted that example, still faced exactly the same response:
```bash
2024-07-06 16:51:52,176 - pysolarmanv5.pysolarmanv5 - DEBUG - Socket setup completed... <socket.socket fd=3, family=2, type=1, proto=6, laddr=('192.168.1.82', 59200), raddr=('192.168.1.119', 8899)>
2024-07-06 16:51:52,176 - pysolarmanv5.pysolarmanv5 - DEBUG - SENT: a5 17 00 10 45 12 00 a2 72 f3 8e 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 02 1e 00 04 90 77 45 15
2024-07-06 16:51:54,187 - pysolarmanv5.pysolarmanv5 - DEBUG - RECD: a5 10 00 10 15 12 1f a2 72 f3 8e 02 01 3e 0c 4c 00 e5 06 00 00 dc 5b 3d 66 05 00 5e 15
Traceback (most recent call last):
(...)
  File "testsSolarman/basicSyncClientTest.py", line 27, in main
    print(modbus.read_input_registers(register_addr=542, quantity=4))
(...)
  File "pysolarmanv5/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5.py", line 256, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame
```

Nevertheless can you please point me where you found that information, or the keywords you used to search for it? So like that I can try read and maybe attempt other values if that's the case.

Thanks!

githubDante, 2024-07-06:
Hi @ssimoe50,

Search for `sofar hyd 3600-es modbus rtu`. The first result which I see points to this [PDF](https://community.openenergymonitor.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf). The addresses seems to be used [in this repo](https://github.com/AndyWhittaker/HYD6000). It's a different model so the mapping may be different.

Another PDF can be found [here](https://solaranzeige.de/phpBB3/download/file.php?id=5254)

Good luck.

githubDante, 2024-08-13:
@ssimoe50 seems to have found the solution for this inverter model. See [here](https://github.com/davidrapan/ha-solarman/issues/70#issuecomment-2285260640) for more info

closed 2024-08-13 by githubDante