jmccrohan/pysolarmanv5 #72: Scanner no longer working

davidlekve, 2024-10-04:
The solarman_scan worked before, but it has now stopped working.  Therefore it seems that there has been a software update to the solarman data loggers that makes the scanning function defect. It might be that it e.g no longer responds to "WIFIKIT-214028-READ".

The read and write to the ip of the data logger still works, but the scan can not find the IP. This shows that there is a data logger on the local network.

Anyone else having the same issue?

davidrapan, 2024-10-04:
Hi @davidlekve,

Yes, the first firmware I know of that has this feature removed is MW3_SSL_5408_1.0B. The binary contains no trace of `WIFIKIT-214028-READ` or any similar string.

davidlekve, 2024-10-04:
Just for information, is this the Module Version I have been having trouble with: LSW3_32U_5406_1.06

Thank you for reply @davidrapan. Do you know if there is a good way to solve this and get the ip for these data loggers?

davidrapan, 2024-10-04:
Hmm that's weird, `LSW3_32U_5406_1.06` is quite old and discovery works with this one (I have device with this version).

githubDante, 2024-10-04:
Does it respond to direct requests?

```python
def test_logger_direct(local_ip: str, logger_ip: str):
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    
    sock.settimeout(1.0)
    request = "WIFIKIT-214028-READ"
    sock.bind((local_ip, 48899))
    address = (logger_ip, 48899)
    sock.sendto(request.encode(), address)

    try:
        data = sock.recv(1024)
        values = data.decode().split(",")
        print(values)
    except socket.timeout:
        pass
```

githubDante, 2024-10-04:
> Hi @davidlekve,
> 
> Yes, the first firmware I know of that has this feature removed is MW3_SSL_5408_1.0B. The binary contains no trace of `WIFIKIT-214028-READ` or any similar string.
@davidrapan  do you have a copy or link to this firmware ?

davidrapan, 2024-10-05:
Hey @davidlekve, I just remembered that it won't respond to the 255.255.255.255. You have to send packet to the broadcast address of particular network e.g. 192.168.0.255 in 192.168.0.0/24.

@githubDante I'm not at home rn I'll look when I get there.

😉

francinv, 2024-10-07:
> Does it respond to direct requests?
>
> ```python
> def test_logger_direct(local_ip: str, logger_ip: str):
>     sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
>
>     sock.settimeout(1.0) [...]

I work with @davidlekve. Thanks for the script. It does respond to direct request:
```python
['WIFIKIT-214028-READ']
```

davidlekve, 2024-10-07:
> Hey @davidlekve, I just remembered that it won't respond to the 255.255.255.255. You have to send packet to the broadcast address of particular network e.g. 192.168.0.255 in 192.168.0.0/24.
> 
> @githubDante I'm not at home rn I'll look when I get there.
> 
> 😉

We have done this and the broadcast address is set to the one of the network.

francinv, 2024-10-07:
When running the [scan script](https://github.com/jmccrohan/pysolarmanv5/blob/main/utils/solarman_scan.py) this block throws an exception:
```python
data = sock.recv(1024)
```

davidrapan, 2024-10-07:
> is set to the one of the network.

You mean to the network to which the inverter belongs to, right?

francinv, 2024-10-07:
> > is set to the one of the network.
> 
> You mean to the network to which the inverter belongs to, right?

Answering for @davidlekve here. 
Yes. The inverter has the IP: `192.168.0.249`, so we use the broadcast address: `192.168.0.255`.

davidrapan, 2024-10-07:
Yeah, just checking.

Then I'm out of ideas and you are proly stuck with probing whole range using unicast...

githubDante, 2024-10-08:
@davidlekve & @francinv, as #73 has been merged you can update with `git pull` and use the unciast scanner available in the `utils/` directory.

closed 2024-10-08 by githubDante