jmccrohan/pysolarmanv5 #75: Socket timeouts and socket closed

francinv, 2024-10-15:
### Background 
I have two Python programs that runs on a RPi:
1. A worker that runs every second and fetches data from the inverter with Modbus/TCP. I only initialize a connection one time.
2. A SellProgrammer that is initialized every hour. I call `programmer.disconnect()`after each update.

### The problem
In the start this works, but after some time (over 24 hrs) I start to get `socket.timeout`. As the time goes by the amount of timeouts increases until it suddenly throws: `Socket closed`. When I stop the service and start it again it however throws: `Connection closed`. 

- Has anyone experienced issues like this before?

**Other questions:**
- Should I disconnect more often when reading?
- Should I store the socket to ensure I use the same one each time?

davidrapan, 2024-10-15:
Hi @francinv, do you instantiate PySolarmanV5/Async w/ `auto_reconnect = True`?

francinv, 2024-10-15:
> Hi @francinv, do you instantiate PySolarmanV5/Async w/ `auto_reconnect = True`?

Yes, I do.

davidrapan, 2024-10-15:
Try to take some inspiration from how I've done it [here](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/api.py). It's pretty resilient.

githubDante, 2024-10-16:
@francinv any not properly closed sockets when this starting to happen ?!? Can you check/provide the output from `netstat` or `ss`

```console 
  ss -nt | grep <logger address>  
```

closed 2024-11-18 by francinv