jmccrohan/pysolarmanv5 #76: The data address received in the request is not an allowable address for the server.

PitinoB, 2024-11-14:
I have LSW3 logger with FW LSW3_15_840B_1.36 on Lenercom hybrid inverter.
solarman_scan.py find logger correctly and print its ip, mac and serial.
When scan registers I got

umodbus.exceptions.IllegalDataAddressError:  The data address received in the request is not an allowable address for the server.

Can enyone help.

jmccrohan, 2024-11-14:
Hi,

Please enable debug mode and show us the request and the response.

Jon

PitinoB, 2024-11-14:
DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=3, family=2, type=1, proto=6, laddr=('192.168.50.31', 38846), raddr=('192.168.50.42', 8899)>
Scanning input registers
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 52 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 30 00 01 2b c9 28 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 52 76 cc 11 46 a6 02 01 5c 30 0c 01 78 10 00 00 18 1c 2a 66 01 84 02 c2 c1 bb 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 53 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 31 00 01 7a 09 b9 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 53 77 cc 11 46 a6 02 01 5c 30 0c 01 78 10 00 00 18 1c 2a 66 01 84 02 c2 c1 bd 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 54 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 32 00 01 8a 09 cb 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 54 78 cc 11 46 a6 02 01 5d 30 0c 01 78 10 00 00 18 1c 2a 66 01 84 02 c2 c1 c0 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 55 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 33 00 01 db c9 de 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 55 79 cc 11 46 a6 02 01 5d 30 0c 01 78 10 00 00 18 1c 2a 66 01 84 02 c2 c1 c2 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 56 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 34 00 01 6a 08 ae 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 56 7a cc 11 46 a6 02 01 5d 30 0c 01 79 10 00 00 18 1c 2a 66 01 84 02 c2 c1 c5 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 57 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 35 00 01 3b c8 41 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 57 7b cc 11 46 a6 02 01 5d 30 0c 01 79 10 00 00 18 1c 2a 66 01 84 02 c2 c1 c7 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 58 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 36 00 01 cb c8 d3 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 58 7c cc 11 46 a6 02 01 5d 30 0c 01 79 10 00 00 18 1c 2a 66 01 84 02 c2 c1 c9 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 59 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 37 00 01 9a 08 e4 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 59 7d cc 11 46 a6 02 01 5d 30 0c 01 79 10 00 00 18 1c 2a 66 01 84 02 c2 c1 cb 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 5a 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 38 00 01 aa 0b f9 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 5a 7e cc 11 46 a6 02 01 5e 30 0c 01 79 10 00 00 18 1c 2a 66 01 84 02 c2 c1 ce 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 5b 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 39 00 01 fb cb 0c 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 5b 7f cc 11 46 a6 02 01 5e 30 0c 01 79 10 00 00 18 1c 2a 66 01 84 02 c2 c1 d0 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 5c 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 3a 00 01 0b cb 1e 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] RECD: a5 13 00 10 15 5c 80 cc 11 46 a6 02 01 5e 30 0c 01 7a 10 00 00 18 1c 2a 66 01 84 02 c2 c1 d3 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] SENT: a5 17 00 10 45 ab 00 cc 11 46 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 75 89 00 01 fa 2c 0c 15
DEBUG:pysolarmanv5.pysolarmanv5:[2789609932] V5_SEQ_NO_MISMATCH: a5 01 00 10 47 aa cf cc 11 46 a6 00 9a 15
DEBUG:pysolarmanv5.pysolarmanv5:Got exception when receiving frame
Traceback (most recent call last):
  File "/root/lenercom_logger/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 280, in _send_receive_v5_frame
    v5_response = self._data_queue.get(timeout=self.socket_timeout)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/queues.py", line 114, in get
    raise Empty
_queue.Empty
Traceback (most recent call last):
  File "/root/pysolarmanv5-main/examples/register_scan.py", line 32, in <module>
    main()
  File "/root/pysolarmanv5-main/examples/register_scan.py", line 15, in main
    val = modbus.read_input_registers(register_addr=x, quantity=1)[0]
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/lenercom_logger/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 568, in read_input_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/lenercom_logger/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 455, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/lenercom_logger/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 422, in _send_receive_modbus_frame
    v5_response_frame = self._send_receive_v5_frame(v5_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/lenercom_logger/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 280, in _send_receive_v5_frame
    v5_response = self._data_queue.get(timeout=self.socket_timeout)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/queues.py", line 114, in get
    raise Empty
_queue.Empty

githubDante, 2024-11-14:
This seems to be expected. The scanner is just an example. You should consult the documentation for the addresses supported by the inverter.