jmccrohan/pysolarmanv5 #87: Error while connecting to the inverter

gummiangler, 2025-02-07:
SN: 902000501
Model: Deye SUN-M80G3-EU-Q0 800W

Cannot connect to the datalogger: 

`pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame contains incorrect data logger serial number`

githubDante, 2025-02-07:
You should use the serial number of the datalogger not the inverter.

gummiangler, 2025-02-07:
> You should use the serial number of the datalogger not the inverter.

I did of course

[image]

githubDante, 2025-02-07:
Can you provide the debug output -> `PysolarmanV5(<address>, <serial>, <other options>, verbose=True)`

gummiangler, 2025-02-08:
@githubDante 
> Can you provide the debug output -> `PysolarmanV5(<address>, <serial>, <other options>, verbose=True)`

Sure!

```
DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=1224, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.178.128', 59598), raddr=('192.168.178.103', 8899)>
DEBUG:pysolarmanv5.pysolarmanv5:[902000501] SENT: a5 17 00 10 45 2d 00 75 6f c3 35 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 80 fe 00 06 38 38 70 15
DEBUG:pysolarmanv5.pysolarmanv5:[902000501] V5_MISMATCH: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 33 32 2d 44 31 0d 0a 0d 0a
DEBUG:pysolarmanv5.pysolarmanv5:[902000501] RECD: a5 10 00 10 15 2d 55 79 a7 e8 ed 02 01 4f dc 36 00 6f 13 00 00 52 8b 70 67 06 00 4c 15
```

davidrapan, 2025-02-08:
Hi @gummiangler, decoded serial from received frame is `3991447417`... on top of that it responded w/ modbus exception code **0x06: Slave Device Busy**

gummiangler, 2025-02-08:
Hi @davidrapan 

Thank you very much! I also wanted to use your Home Assistant integration, but of course I'm getting the same problems. Do you have any suggestions as to what I could do?

githubDante, 2025-02-08:
Use the serial number reported by the datalogger `3991447417` and you should be OK.

gummiangler, 2025-02-08:
@githubDante @davidrapan 
Thank you very much! Now it works

closed 2025-02-08 by gummiangler

davidrapan, 2025-02-08:
@gummiangler, that's why I told you on discord that we needed to see the debug logs.. 😆 I'm glad you got it resolved. 😉