jmccrohan/pysolarmanv5 #90: Can't even Import pysolarmanv5. ImportError: cannot import name 'PySolarmanV5' from 'pysolarmanv5'

Bignose3, 2025-04-08:
Please help, been using HA to control my Solis Inverter, works mostly but HA in on a NAS & running python & jsut seems to need attention a fair bit, modules seem to need reinstalling from time to time & via remote terminals, testing python code is just not great when short of time.
Would prefer most interation via python runing on my PC.
I get solar output from OpenMeteo & adjust the battery charge start time accordingly all in python (on the NAS) but then send via HA , few to many automations & passing of values. Would be so much easier just one python program on a PC seperate from HA.

I know serial below is wrong, will adjust once I get past the basics.

Windows 10
Python 3.13.1 (tags/v3.13.1:0671451, Dec  3 2024, 19:06:28) [MSC v.1942 64 bit (AMD64)] on win32
pip install all seemed to go OK.

""" A basic client demonstrating how to use pysolarmanv5."""

I # (rem'd) out the others when testing, so only one import at a time

1)
from pysolarmanv5 import PySolarmanV5
ImportError: cannot import name 'PySolarmanV5' from 'pysolarmanv5' (consider renaming 'C:\\Users\\Gregory Reed\\LPR\\OpenMeteo\\pysolarmanv5.py' if it has the same name as a library you intended to import)

2)
from pysolarmanv5.pysolarmanv5 import PySolarmanV5
ModuleNotFoundError: No module named 'pysolarmanv5.pysolarmanv5'; 'pysolarmanv5' is not a package

3)
import pysolarmanv5
No error on import but 
'module' object is not callable on ... 
modbus = pysolarmanv5("192.168.2.240", 123456789, port=8899, mb_slave_id=1, verbose=False)

def main():

    """Create new PySolarman instance, using IP address and S/N of data logger

    Only IP address and S/N of data logger are mandatory parameters. If port,

    mb_slave_id, and verbose are omitted, they will default to 8899, 1 and 0

    respectively.

    """

    modbus = pysolarmanv5("192.168.2.240", 123456789, port=8899, mb_slave_id=1, verbose=False)

githubDante, 2025-04-08:
Hi,

Something is wrong with your environment. I'm not very familiar with Windows, but it's always better to create a `virtualenv` , activate it and use `pip install` after that. The proper way to import PySoalrman is the variant 1 –>  `from pysolarmanv5 import PySolarmanV5`

Bignose3, 2025-04-08:
You're right, something odd with my python install, fine with numerous other programs I have but uninstalled, reinstalled & messed about a lot, environment path & some other windows settings not seen before. Eventually pysolarmanv5 just started loading about an hour ago, really not sure what did in the end.
Now the fun part, cannot seem to get it to read the modbus, early days so lots to investigate but have a feeling it might not like my Solis 4 pin ethernet data logger. Have the IP & serial etc. (works with HA) so more lots to test. Does it have to be a solarman or particular data logger?

jmccrohan, 2025-04-08:
Hi @Bignose3,

Glad you got your python environment sorted. Your data logger must be a Solarman-based data logger - DLS-W type. Newer models are not supported.

Cheers,
Jon

davidrapan, 2025-05-04:
@Bignose3, you don't need to write your custom script, there are solarman integrations available in the HACS (or [here](https://github.com/davidrapan/ha-solarman)) which are using pysolarmanv5. 😉